leaflabs/maple-ide #89: Source code for the Windows Drivers

rogerclarkmelbourne, 2015-01-18:
Only binary files seem to be present for the Windows drivers.

Is there source code for these, or were they acquired in binary only form.

If they are not created by Leaflabs, could you tell me they were sourced from, so I can contact the original author

rogerclarkmelbourne, 2015-01-19:
I've since been informed that the drivers are from Microsoft. Could you confirm this?

mbolivar, 2015-01-19:
Hi Roger,

Can you please point me at the files you're talking about?

It's been a long time, but I think we only have a .inf, which, as you say,
is a text file that just requests that Windows uses its built-in virtual
COM port driver.

Thanks,
Marti

rogerclarkmelbourne, 2015-01-19:
Marti,

Thanks for the reply.

The files in https://github.com/leaflabs/maple-ide/tree/master/build/windows/dist/drivers/mapleDrv

Windows 7 and Windows 8 (and 8.1) etc are proving to be problematic with the drivers as Windows 7 onwards requires signed drivers (binaries) and apparently Windows 8.1 requires the whole driver package to be signed somehow including the inf file even if we use the standard / pre-installed  windows usb serial driver.

Someone (tim schuerewegen )  has done a modified version of this repo https://github.com/pbatard/libwdi  which is supposed to install the Windows USB Serial (and possibly DFU drivers - I'm not sure...) onto the VID / PID for the Maple / Maple mini, but I have not had any success getting it to work.

So I just wanted to check whether the driver was indeed the standard Windows usb serial driver, and also if possible to find out about the DFU driver.

PS. 
I presume you picked the VID 0x1EAF because it spelled LEAF, but it doesn't seem to appear in any official VID list that I can find.
I presume you use an unofficial VID.

mbolivar, 2015-01-19:
Hi Roger,

Yes, those directories in your link contain only .inf's, which are text
files, not driver binaries.

We're not planning on supporting later versions of Windows at this time.
There are some workarounds in the IDE installation documentation on how to
disable driver signing if you want to use Maple IDE in those environments.

Thanks,
Marti

rogerclarkmelbourne, 2015-01-19:
Marti

Thanks

I will work with Tim to get a DWI installer exe to install the com driver etc on Windows 8.1  thanks

Roger

closed 2015-01-19 by rogerclarkmelbourne